WasatchPhotonics/ENLIGHTEN #47: "Did you know...?"

mzieg, 2022-10-07:
Add new MessageBox based DidYouKnowFeature, disconnected from current Marquee-based Guide, which _at launch_ informs user of a random high-value but “often missed” feature. Dialog should include opt-out checkbox to “disable tips” (persist), “next tip” (new random draw, unequal to prev), and “learn more” (link to appropriate section in online manual).

Mark can provide list of tips. Biggest question is honestly how to persistently link to _sections_ (not page numbers) of constantly updated PDF.

*sourced from Issue-20220927-4*

samiebee43, 2023-06-26:
> Biggest question is honestly how to persistently link to sections (not page numbers) of constantly updated PDF.

This seems like a good reason to export our manual to HTML format. Is there any reason we've chosen not to do so in the past?

example:
https://wasatchphotonics.com/documents/enlighten-manual-4.x-beta/#h.w3pjtnvn6rdn

mzieg, 2023-06-26:
I personally don't think the exported HTML looks nearly as good as the PDF.  
Or are you talking about maintaining the document in HTML directly?
Ultimately, this is a Marketing as well as Engineering discussion, and not suitable for a Git issue.

samiebee43, 2023-10-16:
Is this blocked in any way?

One way to organize this would be to create a Msgbox class with (at least) three descendants.

1. one to be used for the [stylized](https://github.com/WasatchPhotonics/ENLIGHTEN/blob/ca8b2e4b6a1b02b14abee3ae2e3b3de8e4abe36f/enlighten/common.py#L159) `common.msgbox` function
2. one for the "Are you sure you want to exit Enlighten?" prompt and possibly for a generic `common.msgboxConfirm` function
3. one for this "Did you know...?" prompt

[image]
"Are you sure you want to exit?" prompt with question-mark icon

mzieg, 2023-10-16:
I don't think the pop-up messagebox aspect of it is blocked.  I just don't see a way to maintain links into sections of an automatically exported PDF unless Google provides a way to set bookmark ID (which they currently don't).

We could do this if we maintained the manual in HTML, but I don't want to. We could do this if we manually edited PDF exports using a commercial PDF editor to "forcibly apply" important bookmarks to specific sections (probably only 1-2 dozen such links needed, wouldn't actually be that hard). Or we could probably do this if we moved the Google Doc manual from our ERP drive to a public URL.  That could be the simplest compromise.  Maybe I'm missing something?

samiebee43, 2023-10-17:
I have zero preference how this is done--but I'm leaning towards options that don't generate recurring work. Here I show some more detail about the Google Docs options. Can you elaborate about commercial pdf editors? Can they link to specific sections or specific pages? I think linking to a specific page might be good enough if that's an easy option.

## Google Document options

[image]

### Anyone with Link

If we use Google Doc > Share > Anyone with Link then we get a result [like this one](https://docs.google.com/document/d/1F1rnIn2LG4ELHms8fhfQJS3OgvuaX6vpPHB0lHLqPIE/edit#heading=h.6ullmef94uf4).

This has the drawback that it shows other users and editors inline with the document:
[image]

This is the option I've seen used by some other open source projects.

### Publish to Web

The result looks [like this](https://docs.google.com/document/d/e/2PACX-1vSUq5BEFCEfWP3OkJQCJF42JmwinodTmVzC8SmWwHkEIJKSFnObgX2I1560o-pKs4pIgmuUTxCwNrZT/pub#h.6ullmef94uf4).
[image]

The "Published using Google Docs" banner looks kind of bad, and it looks like the link is slightly off from the actual target section, but still this option may work.

It updates every 5 minutes, which is kind of annoying if you are trying to make a change and see how it looks in the final output.

mzieg, 2023-10-17:
You know, you're right, page numbers should work fine. And all browsers support adding #page=28 to PDF URLs.

Let's just do that. We can add a help.json file to the distro that maps feature names to page numbers, so we just have to update one file when releasing a new manual. And/or keep help.json on our website so users who have online access enabled can always access the latest page number mapping, even from older programs. 

Or, possibly simplest of all, keep old manual versions online and include manual version in the URL so the page numbers are always the same for that ENLIGHTEN version. When they upgrade to a newer ENLIGHTEN it will have links to a (possibly) newer manual, with appropriate page numbers.

Our website will always provide access to the LATEST manual, but context/sensitive help links will remain "hardcoded" to the manual version in use at the time of that program's release. That's actually a pretty sensible behavior.

samiebee43, 2023-10-18:
Aye to linking to the current revision of manual.

A url scheme would be something like this?

```
/manual/latest
/manual/4.0.18
/manual/4.0.17
/manual/4.0.16
```

We should get the hosted version of the manual ready for this before we implement any changes to Enlighten.

It would be solid if we can use a simple HTML view for this window so it can include gifs/pngs of UI, rich text, plus it's easier to make new pages.

Mockup:
[image]

samiebee43, 2023-10-18:
Maybe we can curl latest upon build?

samiebee43, 2023-10-19:
For reference, ISE has a start up hints window that looks like this.

[image]

mzieg, 2024-02-20:
Resolved in #378

closed 2024-02-20 by mzieg