WasatchPhotonics/ENLIGHTEN #48: Virtual Simulator (real-time "mocks" accessible from UI)

mzieg, 2022-10-07:
From Issue-20221007-1 (this was also raised recently internally):

> Please do consider a virtual device for development. Would be much appreciated for development outside laboratory environment.

I would start with a virtual WP-785X-ILP.

I haven't thought this through, but I'm tentatively picturing a "Virtual Spectrometer" widget, which might offer a "Model" pull-down (WP-785X-F13-ILP).  It would be cool if it also included a "Sample" pull-down (cyclohexane, polystyrene, xenon, argon etc) which provided a basic simulation of what the spectra _would_ look like with that excitation, spectral range, FWHM, integration time, and noise characteristics.

Doing a "proper job" on this would take awhile — this has a lot of interesting corners.

The real nest of snakes would be if we allowed "expert mode" to specify an arbitrary spectral range, excitation etc.  It would be very easy to provide a simulation of a virtual spectrometer that isn't actually possible to make, at least with available detectors, COTS optics etc.  Let's leave that on the backburner for now, and start by allowing selection of a few standard models we know well, and then cautiously grow out from there.

rc14193, 2022-10-10:
What are your thoughts on exposing the current test specs through the GUI using the new factory page? I'm thinking a button on that page to add the test spec and it presents a file dialog and the user selects the right folder. It connects and then there is a remove button as well. This means, to start, only one virtual spectrometer at a time can be connected. There can be some defaults that are shipped with Enlighten and placed in the EnlightenSpectra folder.

mzieg, 2022-10-10:
I think it would be a good start to let us test the functionality internally.  Ultimately, when the feature seems to be working well and doing what we think users would like, then I might add a "✅ Virtual Spectrometer" to the current "Spectrometer" widget (below "Lock," "Hide Others" etc), which if checked exposes a comboBox with all the installed virtual spectrometers.