WasatchPhotonics/ENLIGHTEN #75: Gradient colors?

mzieg, 2022-12-01:
This is an odd one.  Sometimes I take a set of spectra in which one parameter is changing in linear fashion, like integration time stepping from 10-1000ms, or laser power stepping from 450-100mW, or gainDB from 0-30dB etc.  In such cases, our usual graph colorization of "random indexed pre-defined colors" doesn't work well — it's hard to visually map the jumble of colors to the expected steady on-screen progression of intensity.

It would be Really Cool if I could select two thumbnails (start and end of range), then either:

1. select start and end colors (i.e. light blue and dark blue), or
2. select a pre-defined rainbow gradient (e.g. VBGR)

...and have all the thumbnails in between automatically receive a smooth gradient in between.

I'm not sure exactly how I would visualize this on the GUI.

I'm not sure if it would be easier or harder to do from a plug-in.

It would be handy however :-)

mzieg, 2022-12-01:
[image]

rc14193, 2022-12-01:
My initial impression is it would be easier to do from a plugin. Plugins already have access to the clipboard from an earlier request from DB. This would make it easy to make a plugin with a spinbox for start and stop idx. The plugin could then just grab those idxs and plot them with an input gradient. I would think making it part of core features would require more gui modification as there would need to be some way to easily select multiple spectra in the clipboard when right now there isn't a way to select spectra from what I understand.