WasatchPhotonics/ENLIGHTEN #89: Upgrading GUI framework

rc14193, 2023-01-09:
Placing notes here from issues encountered for ARM based Macs.

Our current GUI uses PySide2. There is no package built for this for ARM based Macs on PyPI. This means that we would have to build it ourselves if we wanted to run PySide2, but we would also have to deal with any bugs that arise building it ourselves on ARM Macs. Our other options are:
- Upgrade to PySide 6: No PyPI build for Raspberry Pi
- Upgrade to PyQt5: PyQt is GPL licensed as opposed to the LGPL of PySide, but available for RPi and ARM Mac

These considerations along with bug checking by integration testing of the new framework would be needed.

samiebee43, 2023-03-02:
Some background: https://www.pythonguis.com/faq/pyqt5-vs-pyside2/
PySide and PyQt are separate implementations of Qt for Python. 

PyQt has been around for longer, but PySide has been dubbed the official one by Qt.

mzieg, 2024-05-23:
resolved in 4.0.20

closed 2024-05-23 by mzieg