WasatchPhotonics/ENLIGHTEN #97: Enlighten freezes up after repeated device reconnects

samiebee43, 2023-02-02:
After connecting/disconnecting device power many times Enlighten has been observed to freeze up, needing a restart.

Sometimes a "Communication lost" message is displayed.

samiebee43, 2023-02-02:
I'm not yet able to reproduce this bug, but on a session with many reconnects I found the Save button stopped working.
Clicking it results in this to stderr:
```
Traceback (most recent call last):
  File "C:\Users\sbee\Projects\ENLIGHTEN\enlighten\VCRControls.py", line 182, in save
    callback()
  File "C:\Users\sbee\Projects\ENLIGHTEN\enlighten\Controller.py", line 1101, in save_current_spectra
    self.measurements.create_from_spectrometer(spec=spec, view=view)
  File "C:\Users\sbee\Projects\ENLIGHTEN\enlighten\Measurements.py", line 237, in create_from_spectrometer
    measurement = self.factory.create_from_spectrometer(
  File "C:\Users\sbee\Projects\ENLIGHTEN\enlighten\MeasurementFactory.py", line 92, in create_from_spectrometer
    measurement = Measurement(
  File "C:\Users\sbee\Projects\ENLIGHTEN\enlighten\Measurement.py", line 319, in __init__
    self.view               = spec.app_state.technique_name
AttributeError: 'SpectrometerApplicationState' object has no attribute 'technique_name'
```

mzieg, 2023-02-20:
Good catch.  I had not seen that one.  Looks like something that appeared in 4.x.

closed 2023-05-16 by samiebee43