WasatchPhotonics/ENLIGHTEN #99: Plugin execution msgbox shows dark in light mode

samiebee43, 2023-02-06:
[image]

AmanSal1, 2023-02-07:
in which file has this plugin warning window designed and coded ?

AmanSal1, 2023-02-07:
i tried looking for it but i didnt found it

AmanSal1, 2023-02-07:
How do we compile this code ? for test ?

AmanSal1, 2023-02-07:
I guess there is some issues with 2 files  :-

1.enlighten/assets/stylesheets/dark)/enlighten.css
2.enlighten/assets/stylesheets/light)/enlighten.css

In PluginController.py there is a function named "display_plugin_warning" which is used to generate the PluginWarning message box . This message box is not styled in any of the above two files thus causing an error that you have mentioned.

FIX:

QMessageBox {
    background-color: rgb(0,0,0);         # This should be included in the dark/ enlighten.css
  }
QMessageBox {
    background-color: rgb(255,255,255);    #This should be included in the light/enlighten.css
}

samiebee43, 2023-02-08:
Hi Aman. Thank you for your suggestions. I tried it out, but it's not yet working for me.
I will look more into it when I have a chance. Until then please be welcome to fork the repository and create a pull request 😁

AmanSal1, 2023-02-08:
can you plzz guide me on how to compile the code so i can test it locally on my machine ?

AmanSal1, 2023-02-08:
its bit confusing for me

samiebee43, 2023-02-22:
Same thing happens with some tooltips (clipboard)
[image]