WasatchPhotonics/ENLIGHTEN #101: Add ✅ Laser Watchdog Enable

mzieg, 2023-02-07:
Currently, the laser watchdog (only supported with Series-XS spectrometers) supports a single spinbox, spinBox_ee_laser_watchdog_sec, with a range (0, 1200sec).

Please add an "enable" checkbox above the spinbox, and change the spinbox range to (3, 1200sec).

If the enable checkbox is OFF, LaserControlFeature should set the watchdog value to 0 sec and HIDE the spinbox.  (I'd rather not disable it, because it would still show a value, and the value would be non-zero, which would be confusing.)

samiebee43, 2023-02-08:
Completed in #102.

Notes. Changing a QTCheckbox state from within StateChanged signal creates undesirable behavior. This is a common use case when checkboxes have cancellable warning messages associated with them. The solution is to use the Clicked signal instead (which does cover "clicking" with the spacebar). I applied this change to both the watchdog and plugin checkboxes. It should be applied anywhere else where there is a checkbox and a warning message.

samiebee43, 2023-02-15:
The watchdog checkbox is now available in beta 4.0.5

closed 2023-02-15 by samiebee43