WasatchPhotonics/ENLIGHTEN #107: plot right click, set axis persist

samiebee43, 2023-02-22:
Plot window configuration options are presented and hidden based on mouse movement alone. When setting the highlighted field, for instance, it is common for the mouse to hover slightly outside of the context menu causing it to close.

[image]

mzieg, 2023-02-22:
It honestly didn't occur to me that we could control or improve this.  I would think this could be fixed in [pyqtgraph](https://github.com/pyqtgraph/pyqtgraph) itself, perhaps in a pull request from our fork?

samiebee43, 2023-02-22:
This feature was requested internally. If there's not a way to adjust the context menu without going into the source of pyqtgraph, then one option would be to provide an alternative interface for changing the graph's window.