WasatchPhotonics/ENLIGHTEN #108: Graphs will use dark backgrounds when ENLIGHTEN restarts

samiebee43, 2023-02-22:
This has some feature/bug duality to it. On one hand this allows the user to combine light graphs with dark UI, but the other hand having to restart Enlighten to fully enable/disable dark mode is suboptimal.

[image]

This marquee is displayed by GUI.py in `dark_mode_callback`

mzieg, 2023-02-22:
I agree it's messy.  I liked the idea of the light mode, and wanted to keep what we attained there, but didn't have a good solution for updating all the existing graphs and ran out of time (not much was allocated for the theming effort).  Seemed an acceptable compromise to "seal it off."

One thing that would absolutely work is just give each Feature that "owns" a graph, a callback to destroy and recreate the graph in-place (retaining any current curves etc).  I'm pretty sure that would work, just involved more typing than I felt was worth.

I honestly expected light mode to be a "capability demonstration" rather than "daily use" feature. Kind of awaiting user community response on that, after release.  But in any case, I guess I assume that everyone will prefer one mode over another, and just leave it that way, so the once-per-computer relaunch isn't a biggie.

samiebee43, 2023-10-13:
Main graph changes color without restart now (as of sbee-style), but some of the smaller graphs still don't update until restart.

[image]