WasatchPhotonics/ENLIGHTEN #109: In-app shortcut cheatsheet

samiebee43, 2023-02-22:
Working on issue #88 I discovered some of the existing shortcuts for the first time.

For instance Ctrl+1 and Ctrl+2 switch between Scope and Settings. Knowing these shortcuts would have been very helpful for plugin development which often involves frequent switching between these views.

I think it would make sense to advertise these shortcuts to our users via the GUI itself. Here's how some other programs accomplish this.

Google drive
[image]
CMD+/ (mac) or "Help>Keyboard shortcuts" reveals the shortcut actions.

Visual Studio Code
[image]
Context menus show shortcut keys

Two propose ways to do it in Enlighten.

(1) Show shortcut combos next to buttons
[image]

(2) Use dedicated Marquee space (to be added in #92) for a shortcuts cheatsheet button.
[image]
Clicking this button would present all of the shortcuts at once.

samiebee43, 2023-02-22:
[image]
 Some buttons have tooltips as @mzieg pointed out.
Perhaps these can be added to the Scope,Settings,Hardware,Log dropdown

mzieg, 2023-02-22:
> Show shortcut combos next to buttons
> shortcuts cheatsheet button

I like 'em both :-)  All about supporting multiple learning styles...

mzieg, 2023-06-29:
[image]
[image]
[image]

mzieg, 2023-06-29:
also addressed in #214

closed 2024-12-02 by mzieg