WasatchPhotonics/ENLIGHTEN #111: Enlighten CSV loader is broken in 4.0.x

samiebee43, 2023-02-23:
This error is independent of the CSV file itself. It looks like MeasurementFactory referencees a new field called 'view' but Measurement has not got the memo to include such a field

```
2023-02-23 16:55:50,270 [0x0000057c] enlighten.MeasurementFactory ERROR    create_from_file: error finalizing measurement
Traceback (most recent call last):
  File "C:\Users\sbee\Projects\ENLIGHTEN\enlighten\MeasurementFactory.py", line 276, in create_from_file
    self.create_thumbnail(m, is_collapsed)
  File "C:\Users\sbee\Projects\ENLIGHTEN\enlighten\MeasurementFactory.py", line 125, in create_thumbnail
    view            = measurement.view,
AttributeError: 'Measurement' object has no attribute 'view'
2023-02-23 16:55:50,270 [0x0000057c] enlighten.Plugins.PluginController DEBUG    received Measurement from event load
2023-02-23 16:55:50,270 [0x0000057c] enlighten.Measurements DEBUG    create_from_file: completing new measurement
2023-02-23 16:55:50,270 [0x0000057c] enlighten.Measurements ERROR    unable to add Measurement w/o ThumbnailWidget
```

mzieg, 2023-02-24:
Context: probably broke in 4.x when we renamed the old "Technique" menu to "View."  "Technique" is still the correct name for the field in saved files, so that should be switched back in MeasurementFactory.

ENLIGHTEN <4.x had a single menu in which genuine spectroscopic techniques like Raman, Transmission/Reflectance, Absorbance etc were weirdly intermixed with GUI features more appropriately named "Views" like Scope and Hardware.

Separating those into two different comboBoxes in 4.x was the right decision; it looks like we went a little overboard with a batch rename in this case.

samiebee43, 2023-02-27:
fixed 526a30b