WasatchPhotonics/ENLIGHTEN #112: Enlighten CSV Loader missing wavecal

samiebee43, 2023-02-27:
Measurement is being instantiated with a `source_pathname`, which means `settings` is passed in by the caller. At the time of instantiation the settings variable already does not have wavecal info...

`settings.state.region -> None`
`settings.eeprom.wavelength_coeffs -> []`

Note that the live spectrometer _does_ have wavecal, it's just not being saved (or not being parsed) from CSV.

Places that call Measurement and define settings are as follows:

- ColumnFileParser
- DashFileParser
- TextFileParser

My file does not call TextFileParser when it is loaded. It is yet to be narrowed down to ColumnFileParser or DashFileParser.

[Session-20230227-122814.csv](https://github.com/WasatchPhotonics/ENLIGHTEN/files/10844707/Session-20230227-122814.csv)

samiebee43, 2023-04-04:
Further diving into FileParsers has that this file is mostly like used by ColumnFileParser (if not ExportFileParser which wasn't in the list).

The settings parameter has been involved with the refactor, if this is an unbound variable issue it may already be fixed.

mzieg, 2023-06-29:
I think this may have been fixed in #168? I was able to load your session file and it "looked" like the wavenumber and wavelength axes were being loaded properly.  Please re-test with v4.0.12+, else considered "pending close".

samiebee43, 2023-07-03:
This is what I have in the tooltip on Enlighten 4.0.12 (+ some probably unrelated local changes)

[image]

I think the issue is not loading, actually. Because the file is reporting a 0 wavecal and Enlighten shows the same.

The problem is that at the time of save I had a non-zero wavecal

mzieg, 2023-07-04:
> to check this is resolved, it would make sense to save a new spectra and confirm that wavecal makes it in.

I can confirm that:

[image]

closed 2023-07-04 by samiebee43