WasatchPhotonics/ENLIGHTEN #113: Enlighten CSV utf-8 support

samiebee43, 2023-03-13:


mzieg, 2023-03-13:
Note that we made changes around UTF-8-SIG support in 3.2.8.
```
- 2022-06-06 3.2.8
    - auto-detect utf-8-sig encoding (common in Excel CSV) 
```
https://github.com/WasatchPhotonics/ENLIGHTEN/pull/14

samiebee43, 2023-03-16:
Measurement factory supports `utf-8` and `utf-8-sig` encodings. I've identified that the file causing errors is encoded as `iso8859_2`, which is not fully contained by utf-8.

In b7f037b I've demonstrated how to get it working again, by expanding our `determine_encoding` function slightly.

In truth, encoding detection is a fairly large and vaguely subjective task. We may want to use something like [Chardet](http://chardet.readthedocs.io) which uses a variety of heuristics to make an appropriate encoding guess.