WasatchPhotonics/ENLIGHTEN #132: Plugin Baseline Enhancements

samiebee43, 2023-04-25:
- [ ] Center btn --> move plugin.x to vertex-space y-max, gradient ascent
- [ ] Use Enlighten.xcursor for plugin.x
- [ ] pandas table --> Measurement/CSV
- [ ] run plugin on CSV
- [ ] Use trapezoidal integration
- [ ] Use parabolic estimation for peak height
- [ ] Consider using vertex U {parabolic} for integration

samiebee43, 2023-05-08:
2023-05-08

- [ ] independent left and right outer radii (blue)
- [ ] remote inner radius (green)
- [ ] remove decimals from pandas frame
- [ ] add commas to pandas frame
- [ ] Use spectra intersection outer radius + linear interpolation for baseline
---
- [ ] resizable pandas table
- [ ] move plugin to top of sidebar

mzieg, 2023-05-08:
I'm unsure if moving to the "very top" of the sidebar is ideal.  Personally I might vote for between (Detector Control, X-Axis) or (X-Axis, Scan Averaging).  I think laser control and detector control (integration time) should remain at the top though.  Making it harder to find (unexpectedly moving) the button to turn off the laser is almost a safety issue.

samiebee43, 2023-05-16:
Moved checklist items to first comment so this thing is accurate:
[image]

samiebee43, 2023-05-18:
Was having issues were logging wasn't working. Turns out spec needs to be on for "Process" btn to do anything. Led to the addition of plugins having their own log file, which seems overdue anyway

samiebee43, 2023-06-09:
Some notes about the current state of this.

When LocalBaseline is enabled and a spec is connected, changing the count clears the widgets.
[image]
[image]

This signifies that immediate-mode widget onchange detection is working and post-config widget spawning is on its way to working. onchange is linked to `reset_configuration` which I think was intended to recreate the layout. It might just be missing a new call to get_configuration_obj. 

Maintaining field contents can be handled at the plugin level. Currently it is only for the `count` parameter which must be persistent for multiplexing to work at all. Persistence of the other fields is an (important) usability improvement.

closed 2023-06-15 by samiebee43