WasatchPhotonics/ENLIGHTEN #133: RamanShiftCorrectionFeature find_peaks FWHM

mzieg, 2023-04-26:
The 'width' parameter in RamanShiftCorrectionFeature's call to scipy.signal.find_peaks should probably be tied to Spectrometer.fwhm rather than a hardcoded constant for all spectrometers.  (The current implementation has apparently been found to work on all tested units, but it still seems like a design weakness.)

mzieg, 2023-06-29:
Internal user reported that RamanShiftCorrection was failing on some X-Series units during characterization...this could well be related. Bumping priority.

samiebee43, 2023-07-17:
Can we get more info about how RamanShiftCorrection is failing?

samiebee43, 2023-07-17:
And can we make this into a separate issue? You can link them by referencing the issue number. That way I can solve the above reimplement without shadowing a potentially still active bug.

PR #289 (2023-09-20): Reference spec.fwhm instead of using default value 3
Close #133
commit 3ac241e3 (2023-09-20): Reference spec.fwhm instead of using default value 3

Close #133
closed 2023-09-20 by mzieg