WasatchPhotonics/ENLIGHTEN #147: Internet access doesn't always work?

mzieg, 2023-05-12:
I was able to reproduce on user KP's laptop that ENLIGHTEN 4.0.10 was unable to download an Andor config file from AWS, even though internet was otherwise working on the laptop.  Possibly issue with Win11? (I only tested locally on Win10)  Low-priority, but annoying.

samiebee43, 2023-05-15:
Did this have anything to do with `keys.py` and the Network folder?

mzieg, 2023-05-15:
...possibly.  I just verified on my own laptop.  It's not working from the 4.0.10 installer (that I built :-( )

mzieg, 2023-05-15:
I'm not sure this has ever worked through our installers.  Or maybe it did, but it is now broken. I don't think it has anything to do with the keys file.  If keys.py wasn't being compiled in, CloudManager.py wouldn't be trying to connect.

mzieg, 2023-05-15:
Failing transcript (run from binary installer):

```
2023-05-15 11:57:08,440 [0x000019c8] enlighten.Controller DEBUG    attempting to download Andor EEPROM
2023-05-15 11:57:08,440 [0x000019c8] enlighten.network.CloudManager DEBUG    checking if internet available
2023-05-15 11:57:08,472 [0x000019c8] enlighten.network.CloudManager INFO     internet is available
2023-05-15 11:57:08,472 [0x000019c8] enlighten.network.CloudManager DEBUG    creating cloud session
2023-05-15 11:57:08,472 [0x000019c8] enlighten.network.CloudManager DEBUG    instantiating boto3 client
2023-05-15 11:57:08,472 [0x000019c8] botocore.hooks DEBUG    Changing event name from creating-client-class.iot-data to creating-client-class.iot-data-plane
2023-05-15 11:57:08,472 [0x000019c8] botocore.hooks DEBUG    Changing event name from before-call.apigateway to before-call.api-gateway
2023-05-15 11:57:08,472 [0x000019c8] botocore.hooks DEBUG    Changing event name from request-created.machinelearning.Predict to request-created.machine-learning.Predict
2023-05-15 11:57:08,472 [0x000019c8] botocore.hooks DEBUG    Changing event name from before-parameter-build.autoscaling.CreateLaunchConfiguration to before-parameter-build.auto-scaling.CreateLaunchConfiguration
2023-05-15 11:57:08,472 [0x000019c8] botocore.hooks DEBUG    Changing event name from before-parameter-build.route53 to before-parameter-build.route-53
2023-05-15 11:57:08,472 [0x000019c8] botocore.hooks DEBUG    Changing event name from request-created.cloudsearchdomain.Search to request-created.cloudsearch-domain.Search
2023-05-15 11:57:08,472 [0x000019c8] botocore.hooks DEBUG    Changing event name from docs.*.autoscaling.CreateLaunchConfiguration.complete-section to docs.*.auto-scaling.CreateLaunchConfiguration.complete-section
2023-05-15 11:57:08,472 [0x000019c8] botocore.hooks DEBUG    Changing event name from before-parameter-build.logs.CreateExportTask to before-parameter-build.cloudwatch-logs.CreateExportTask
2023-05-15 11:57:08,472 [0x000019c8] botocore.hooks DEBUG    Changing event name from docs.*.logs.CreateExportTask.complete-section to docs.*.cloudwatch-logs.CreateExportTask.complete-section
2023-05-15 11:57:08,472 [0x000019c8] botocore.hooks DEBUG    Changing event name from before-parameter-build.cloudsearchdomain.Search to before-parameter-build.cloudsearch-domain.Search
2023-05-15 11:57:08,472 [0x000019c8] botocore.hooks DEBUG    Changing event name from docs.*.cloudsearchdomain.Search.complete-section to docs.*.cloudsearch-domain.Search.complete-section
2023-05-15 11:57:08,487 [0x000019c8] botocore.credentials DEBUG    Looking for credentials via: env
2023-05-15 11:57:08,487 [0x000019c8] botocore.credentials DEBUG    Looking for credentials via: assume-role
2023-05-15 11:57:08,487 [0x000019c8] botocore.credentials DEBUG    Looking for credentials via: assume-role-with-web-identity
2023-05-15 11:57:08,487 [0x000019c8] botocore.credentials DEBUG    Looking for credentials via: sso
2023-05-15 11:57:08,487 [0x000019c8] botocore.credentials DEBUG    Looking for credentials via: shared-credentials-file
2023-05-15 11:57:08,487 [0x000019c8] botocore.credentials DEBUG    Looking for credentials via: custom-process
2023-05-15 11:57:08,487 [0x000019c8] botocore.credentials DEBUG    Looking for credentials via: config-file
2023-05-15 11:57:08,487 [0x000019c8] botocore.credentials DEBUG    Looking for credentials via: ec2-credentials-file
2023-05-15 11:57:08,487 [0x000019c8] botocore.credentials DEBUG    Looking for credentials via: boto-config
2023-05-15 11:57:08,487 [0x000019c8] botocore.credentials DEBUG    Looking for credentials via: container-role
2023-05-15 11:57:08,487 [0x000019c8] botocore.credentials DEBUG    Looking for credentials via: iam-role
2023-05-15 11:57:08,487 [0x000019c8] urllib3.connectionpool DEBUG    Starting new HTTP connection (1): 169.254.169.254:80
  (one sec)
2023-05-15 11:57:09,503 [0x000019c8] urllib3.connectionpool DEBUG    Starting new HTTP connection (2): 169.254.169.254:80
  (one sec)
2023-05-15 11:57:10,504 [0x000019c8] botocore.loaders DEBUG    Loading JSON file: C:\Program Files\Wasatch Photonics\ENLIGHTEN\Enlighten\botocore\data\endpoints.json
2023-05-15 11:57:10,519 [0x000019c8] botocore.loaders DEBUG    Loading JSON file: C:\Program Files\Wasatch Photonics\ENLIGHTEN\Enlighten\botocore\data\sdk-default-configuration.json
2023-05-15 11:57:10,519 [0x000019c8] botocore.hooks DEBUG    Event choose-service-name: calling handler <function handle_service_name_alias at 0x0000026F47C5E0E0>
2023-05-15 11:57:10,597 [0x000019c8] botocore.loaders DEBUG    Loading JSON file: C:\Program Files\Wasatch Photonics\ENLIGHTEN\Enlighten\botocore\data\cognito-identity\2014-06-30\service-2.json
2023-05-15 11:57:10,675 [0x000019c8] botocore.loaders DEBUG    Loading JSON file: C:\Program Files\Wasatch Photonics\ENLIGHTEN\Enlighten\botocore\data\cognito-identity\2014-06-30\endpoint-rule-set-1.json.gz
2023-05-15 11:57:10,675 [0x000019c8] botocore.loaders DEBUG    Loading JSON file: C:\Program Files\Wasatch Photonics\ENLIGHTEN\Enlighten\botocore\data\partitions.json
2023-05-15 11:57:10,690 [0x000019c8] botocore.hooks DEBUG    Event creating-client-class.cognito-identity: calling handler <function add_generate_presigned_url at 0x0000026F47BCD990>
2023-05-15 11:57:10,690 [0x000019c8] botocore.endpoint DEBUG    Setting cognito-identity timeout as (60, 60)
2023-05-15 11:57:10,690 [0x000019c8] enlighten.network.CloudManager DEBUG    getting client credentials
[... cut ...]
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\mzieg\\AppData\\Local\\Temp\\tmpzd_x20yy'
2023-05-15 11:57:10,690 [0x000019c8] botocore.hooks DEBUG    Event needs-retry.cognito-identity.GetId: calling handler <bound method RetryHandler.needs_retry of <botocore.retries.standard.RetryHandler object at 0x0000026F421BD030>>
2023-05-15 11:57:10,690 [0x000019c8] botocore.retries.standard DEBUG    Retry needed, retrying request after delay of: 0.32240068155206736
```

mzieg, 2023-05-15:
Working transcript (run from source):
```
2023-05-15 12:12:32,511 [0x00000854] enlighten.Controller DEBUG    attempting to download Andor EEPROM
2023-05-15 12:12:32,511 [0x00000854] enlighten.network.CloudManager DEBUG    checking if internet available
2023-05-15 12:12:32,574 [0x00000854] enlighten.network.CloudManager INFO     internet is available
2023-05-15 12:12:32,574 [0x00000854] enlighten.network.CloudManager DEBUG    creating cloud session
2023-05-15 12:12:32,574 [0x00000854] enlighten.network.CloudManager DEBUG    instantiating boto3 client
2023-05-15 12:12:32,574 [0x00000854] botocore.hooks DEBUG    Changing event name from creating-client-class.iot-data to creating-client-class.iot-data-plane
2023-05-15 12:12:32,574 [0x00000854] botocore.hooks DEBUG    Changing event name from before-call.apigateway to before-call.api-gateway
2023-05-15 12:12:32,574 [0x00000854] botocore.hooks DEBUG    Changing event name from request-created.machinelearning.Predict to request-created.machine-learning.Predict
2023-05-15 12:12:32,574 [0x00000854] botocore.hooks DEBUG    Changing event name from before-parameter-build.autoscaling.CreateLaunchConfiguration to before-parameter-build.auto-scaling.CreateLaunchConfiguration
2023-05-15 12:12:32,574 [0x00000854] botocore.hooks DEBUG    Changing event name from before-parameter-build.route53 to before-parameter-build.route-53
2023-05-15 12:12:32,574 [0x00000854] botocore.hooks DEBUG    Changing event name from request-created.cloudsearchdomain.Search to request-created.cloudsearch-domain.Search
2023-05-15 12:12:32,574 [0x00000854] botocore.hooks DEBUG    Changing event name from docs.*.autoscaling.CreateLaunchConfiguration.complete-section to docs.*.auto-scaling.CreateLaunchConfiguration.complete-section
2023-05-15 12:12:32,574 [0x00000854] botocore.hooks DEBUG    Changing event name from before-parameter-build.logs.CreateExportTask to before-parameter-build.cloudwatch-logs.CreateExportTask
2023-05-15 12:12:32,574 [0x00000854] botocore.hooks DEBUG    Changing event name from docs.*.logs.CreateExportTask.complete-section to docs.*.cloudwatch-logs.CreateExportTask.complete-section
2023-05-15 12:12:32,574 [0x00000854] botocore.hooks DEBUG    Changing event name from before-parameter-build.cloudsearchdomain.Search to before-parameter-build.cloudsearch-domain.Search
2023-05-15 12:12:32,574 [0x00000854] botocore.hooks DEBUG    Changing event name from docs.*.cloudsearchdomain.Search.complete-section to docs.*.cloudsearch-domain.Search.complete-section
2023-05-15 12:12:32,589 [0x00000854] botocore.credentials DEBUG    Looking for credentials via: env
2023-05-15 12:12:32,589 [0x00000854] botocore.credentials DEBUG    Looking for credentials via: assume-role
2023-05-15 12:12:32,589 [0x00000854] botocore.credentials DEBUG    Looking for credentials via: assume-role-with-web-identity
2023-05-15 12:12:32,589 [0x00000854] botocore.credentials DEBUG    Looking for credentials via: sso
2023-05-15 12:12:32,589 [0x00000854] botocore.credentials DEBUG    Looking for credentials via: shared-credentials-file
2023-05-15 12:12:32,589 [0x00000854] botocore.credentials DEBUG    Looking for credentials via: custom-process
2023-05-15 12:12:32,589 [0x00000854] botocore.credentials DEBUG    Looking for credentials via: config-file
2023-05-15 12:12:32,589 [0x00000854] botocore.credentials DEBUG    Looking for credentials via: ec2-credentials-file
2023-05-15 12:12:32,589 [0x00000854] botocore.credentials DEBUG    Looking for credentials via: boto-config
2023-05-15 12:12:32,589 [0x00000854] botocore.credentials DEBUG    Looking for credentials via: container-role
2023-05-15 12:12:32,589 [0x00000854] botocore.credentials DEBUG    Looking for credentials via: iam-role
2023-05-15 12:12:32,589 [0x00000854] urllib3.connectionpool DEBUG    Starting new HTTP connection (1): 169.254.169.254:80
2023-05-15 12:12:33,589 [0x00000854] urllib3.connectionpool DEBUG    Starting new HTTP connection (2): 169.254.169.254:80
2023-05-15 12:12:34,605 [0x00000854] botocore.loaders DEBUG    Loading JSON file: C:\Users\mzieg\miniconda3\envs\conda_enlighten3\lib\site-packages\botocore\data\endpoints.json
2023-05-15 12:12:34,605 [0x00000854] botocore.loaders DEBUG    Loading JSON file: C:\Users\mzieg\miniconda3\envs\conda_enlighten3\lib\site-packages\botocore\data\sdk-default-configuration.json
2023-05-15 12:12:34,605 [0x00000854] botocore.hooks DEBUG    Event choose-service-name: calling handler <function handle_service_name_alias at 0x00000271E8065FC0>
2023-05-15 12:12:34,652 [0x00000854] botocore.loaders DEBUG    Loading JSON file: C:\Users\mzieg\miniconda3\envs\conda_enlighten3\lib\site-packages\botocore\data\cognito-identity\2014-06-30\service-2.json
2023-05-15 12:12:34,714 [0x00000854] botocore.loaders DEBUG    Loading JSON file: C:\Users\mzieg\miniconda3\envs\conda_enlighten3\lib\site-packages\botocore\data\cognito-identity\2014-06-30\endpoint-rule-set-1.json.gz
2023-05-15 12:12:34,714 [0x00000854] botocore.loaders DEBUG    Loading JSON file: C:\Users\mzieg\miniconda3\envs\conda_enlighten3\lib\site-packages\botocore\data\partitions.json
2023-05-15 12:12:34,714 [0x00000854] botocore.hooks DEBUG    Event creating-client-class.cognito-identity: calling handler <function add_generate_presigned_url at 0x00000271E7FD5630>
2023-05-15 12:12:34,714 [0x00000854] botocore.endpoint DEBUG    Setting cognito-identity timeout as (60, 60)
[... cut ...] ASUAAJG', 'x-amz-crc32': '755374385'}, 'RetryAttempts': 0}}
2023-05-15 12:12:35,417 [0x00000854] enlighten.Controller DEBUG    andor_eeprom = {'excitation_nm_float': 1063.83, 'detector_serial_number': 'CCD-26826', 'raman_intensity_coeffs': [0.005259877, -0.001453733, 2.027716e-05, -1.015509e-07, 2.142931e-10, -1.533194e-13], 'detector_type': 'iDus', 'wp_serial_number': 'WP-01265', 'wp_model': 'WP-1064XL-F15-XR-IC', 'wavelength_coeffs': [1081.88, 0.6353086, -7.287224e-06, -2.989192e-08, 0.0]}
```

samiebee43, 2023-07-17:
Removing assignment, as I'm taking that to mean "actively in progress". Feel free to readd later

mzieg, 2023-08-09:
Purportedly this did work at the QC station on WP-00918 o_O

mzieg, 2024-02-20:
Seems to work from 4.0.42 installer with WP-01635. Resolving.

closed 2024-02-20 by mzieg