WasatchPhotonics/ENLIGHTEN #150: LocalBaseline wishlist

samiebee43, 2023-05-16:
- [ ] Use Enlighten.xcursor for plugin.x
- [ ] Parabolic estimation peak height
- [ ] Parabolic estimation peak area
- [ ] Run on CSV import

mzieg, 2023-05-16:
> Run on CSV import

Note that there is a setting somewhere for "post-process spectra on load", basically to "load" the Raw column and re-process it using current baseline correction, boxcar, peak sharpening etc...and which *should* automatically trigger a currently enabled plugin.

mzieg, 2023-05-18:
Anecdotal report today that we may have broken area calculation when moving to the independently movable shoulders.

samiebee43, 2023-06-05:
It was intentionally omitted. Independently movable shoulders came with slanted baseline which changes the area calculation. At the time it was deemphasized for the sake of peak height. Adding it back to wishlist

samiebee43, 2023-10-16:
With the buzz about this plugin dying down, and the need for plugin refactoring ever present, I think we should close as 'wontfix' for the time being. This particular wishlist deviates from the end-user's requirements. Other things are more pressing.

closed 2023-10-16 by samiebee43