WasatchPhotonics/ENLIGHTEN #156: Multiple Cursors

mzieg, 2023-05-18:
The addition of "multiplexing" to the BaselinePlugin got me thinking about our whole Cursor feature.  Right now, we've got one Cursor, and you can turn it on or off.  It might be worthwhile just to replace the current [x] enable checkbox with [+] Add cursor, and let the user create as many as they like.  I might move the y-value display from the StatusBar up to the widget, and just show (x, y) tuples for each Cursor (along with a button to assign color).  Any plugin could then access the current list of Cursors and use them to provide on-screen "draggable" x-coordinate inputs to the plugin, which will honestly be a pretty common request.

samiebee43, 2023-05-18:
Love that I could distinguish this from a breaking bug due to our new tags :)

samiebee43, 2023-07-10:
Starting with some designs.

## Existing cursor x/y
x is highlighted orange, y is purple
[image]

## New location
[image]

## Mockup: Cursor default

[image]
## Mockup: Cursor multiple

[image]

## Decisions

We should decide if we still want checkboxes with add/delete. The advantage is that this way keeps track of the user's x-values like the current implementation does for the single cursor. The disadvantage is that it starts to clutter.

[image]

We may also consider using the 'eye-ball'. It's supposed to be a checkbox specific to visibility:

[image]

I'm showing two styles in the same screenshot, borrowed from gimp and blender.

## Enlighten eyeball

After writing the last section, I just noticed we currently use the eyeball in Enlighten, but not exactly correctly. 

[image]

The eyeball is used to close some accordions, but that should be a down arrow button. Likewise the visibility of each spectra is toggled using a button with a spectrum icon-- that should be an eyeball too. The spectrum icon could be better used in the Scope dropdown.

Anyway, the takeaway here is that we have an existing eye-ball icon we can use for multicursor. And while I haven't decided if I want to include them-- I think they are better than checkboxes here.

Rule of thumb, each widget answers a specific type of question, despite being both binary toggles.
checkbox = on or off, enabled or disabled
eyeball = visible or invisble

mzieg, 2023-07-10:
I think I'd vote to ditch the eyeball/checkbox to the left of the x-coord, and just keep the "X remove" button to the right.

I'd consider making it a little table so we could remove "nm/cm⁻¹/px" and "counts" from each row, and just make those column headers.  Also makes it clear that "no, you can't have different cursors with different x/y units."

samiebee43, 2023-07-12:
Here's the latest design iteration.

[image]

We can also sneak something along the lines of (pseudo):

```
on_delete_cursor:
     last_deleted_cursor_x = target.x

on_create_cursor:
    target.x = last_deleted_cursor_x
```

This way the user who was used to enabling/disabling a single cursor can switch to adding/deleting without having to retype the x-value.

mzieg, 2023-07-12:
Very slick!