WasatchPhotonics/ENLIGHTEN #157: Auto-check for updates

mzieg, 2023-05-18:
Basically, if the user has opted-in to internet connectivity in Settings, at launch auto-check:

https://wasatchphotonics.com/binaries/apps/enlighten/sw_versions.json 

...to see if there's a newer version.  If so, prompt to download:

```
ENLIGHTEN 4.0.11 (development) has been posted. You are using 3.2.36 (latest stable). Would you like to update?
(Cancel) (Ignore this version) (Download)
[  ] disable version checking
```

For the record, I'm not sold on the current format of the JSON file...I was actually picturing something more like this:

```
{
  "Applications": {
    "ENLIGHTEN": {
      "Stable": [ "3.2.36", "https://wasatchphotonics.com/binaries/apps/enlighten/Enlighten-Setup64-3.2.36.exe" ],
      "Dev": [ "4.0.11", "https://wasatchphotonics.com/binaries/apps/enlighten/beta/Enlighten-Setup64-4.0.11.exe" ],
    }
  }
}
```

In terms of persistence, we'd just be saving the list of versions they said not to bug about again, so they'd still get notified of even-newer versions unless they ticked "disable version checking".

mzieg, 2023-07-19:
Note we have a checkbox they have to tick to allow any internet access at all.

samiebee43, 2023-07-20:
> Note we have a checkbox they have to tick to allow any internet access at all.

That is good. I was going to include a note about "avoiding implicit network communication for non-internet application" but I figured it was outside the scope of this topic. I'm glad it was well handled in advance.

mzieg, 2024-05-23:
Looks like this was stubbed but never implemented:

https://github.com/WasatchPhotonics/ENLIGHTEN/blob/main/enlighten/network/UpdateChecker.py