WasatchPhotonics/ENLIGHTEN #161: Reimplement Thumbnail Tooltip Metadata

mzieg, 2023-05-25:
Right now, if you mouseOver a Thumbnail on the Clipboard, it gets a "huge" ToolTip providing all the Metadata for that Measurement. Some users like that, others vocally do not.

One option would be to add a button to the "Thumbnail button bar" under each Thumbnail, where clicking it would show/hide the metadata.  Or pop it out in a Dialog.

Or maybe we could "ctrl-MouseOver" or something?

Or add a configurable checkbox in Settings?

Or...whatever.  Different users like different things :-)  Just capturing for the backlog...

lenziwill, 2023-05-26:
We should consider ENLIGHTEN default settings, the settings that are active when the software is opened. For example, should ROI and SRM both be on when software is opened?

mzieg, 2024-06-06:
This was changed to WhatsThis.

closed 2024-06-06 by mzieg