WasatchPhotonics/ENLIGHTEN #178: Ubuntu 22 LTS 4.0.11 tarball

samiebee43, 2023-06-09:
Build task.

samiebee43, 2023-06-09:
PySide2 has a [breaking bug](https://bugreports.qt.io/browse/PYSIDE-1226) as of Python 3.8+, and we are now on Python 3.10. The bug is resolved in 5.15.2.1, but PySide2 only has up to version 5.13.2 on Ubuntu 22 LTS.

Workaround attempts:
Use pyside6-uic: Failed due to setMinSize invalid 0,0
Use pyuic5 in rebuild_resources.sh in the place of pyside2-uic

    $ LD_LIBRARY_PATH="~/miniconda3/envs/conda_enlighten3/lib/python3.11/site-packages/PyQt5/Qt5/lib/"

TypeError: 'PySide2.QtWidgets.QWidget.setMinimumSize' called with wrong argument types
Not sure why enlighten_layout.py still has PySide2 references.. 
Tried changing to PySide6 RCC
And changing scripts/Enlighten.py to PyQt5
Tried BasicWindow to PyQt5 (no Signal), and PySide6.

Finally that mixture gives:
```
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
QWidget: Must construct a QApplication before a QWidget
Aborted (core dumped)
```

That was on Wayland. Using X just hides the warning message. (`QT_QPA_PLATFORM=wayland` does nothing.)

---

I mostly wanted to see what would happen if I stuck my hand into the blender that is our multi-library Qt dependency situation... aaaand I think I'll take my chances downgrading conda-linux.yml to Python 3.7 and rebuilding the env. Probably won't be super clean, but I think it'll will be better than this.

mzieg, 2023-06-09:
I've had to do this on RPi to get around broken PySide2:

- run scripts\rebuild_resources.sh on Windows
- rsync enlighten\assets\ from Windows to Linux

samiebee43, 2023-06-13:
Using windows assets still crashes because how the ui file is calling PySide2 (using features from 5.15+).
```
  File "/home/sbee/Projects/Enlighten/enlighten/ui/BasicWindow.py", line 27, in __init__
    self.ui.setupUi(self)
  File "/home/sbee/Projects/Enlighten/enlighten/assets/uic_qrc/enlighten_layout.py", line 2402, in setupUi
    self.formLayout_12.setFormAlignment(Qt.AlignHCenter|Qt.AlignTop)
TypeError: 'PySide2.QtCore.Qt.AlignmentFlag' object cannot be interpreted as an integer
```

Now trying python downgrade...

samiebee43, 2023-06-13:
Python downgrade got it to open.

[image]

Stuck on connecting to device...

Downgrade includes removing novel language features from Enlighten, Wasatch, SpyCWriter. Those features include type annotations, [walrus (ie x := 5)](https://docs.python.org/3/whatsnew/3.8.html#assignment-expressions), and dataclasses.

Each of these repos should have a ubu-py37 branch from which we'll cut the next linux release. SpyCWriter is added as a local import.

samiebee43, 2023-06-15:
Tried running `demo-simple.py` from Wasatch.PY ubu-py37, and found that "SpectrometerResponse takes no arguments". The solution to that was to expand the dataclasses. Now `demo-simple.py` runs completely, and Enlighten flashes a green graph for a split second before settling on a connected + legend state W/O Spectra:
[image]

EEPROM reading is fine
[image]

Latest hint from enlighten.log
```
error change_setting() got an unexpected keyword argument 'args' with trying to set setting num_connected_devices with args and kwargs () and {'args': [1]}
```

samiebee43, 2023-06-15:
Further iterations on that showed that further purging of dataclasses was needed.
[image]

Display window ✅
Read EEPROM ✅
Show Spectra ✅
X-Cursor ✅
X-Axis  ✅
Set integration time  ✅
Scan Averaging  ✅
Boxcar  ✅
Plugins ❌ (crash, inner-frame frozen, logs continuing)
Save spectra ✅
Load spectra ❌ (quietly fail to load)
Write EEPROM ✅

samiebee43, 2023-06-15:
Fixed load spectra. [Uploaded .tgz to beta site](https://wasatchphotonics.com/binaries/apps/enlighten/beta/ENLIGHTEN-Ubuntu22-4.0.12.tgz).

samiebee43, 2023-06-19:
Device support is present running from source (following instructions in LINUX_README.md). Clone Enlighten, Wasatch, SpyCWriter, and checkout ubu-py37 on all.

It's not working via the 'installed' version which is stuck on libusb-1.0 for some reason. The runtime produces an error caused by libusb0.py using libusb-1-*.so as a (fallback) backend.
```
Traceback (most recent call last):                                                                                                                                                                         
  File "wasatch/WasatchBus.py", line 52, in update                                                                                                                                                         
  File "wasatch/DeviceFinderUSB.py", line 129, in find_usb_devices                                                                                                                                         
  File "wasatch/DeviceFinderUSB.py", line 47, in bus_polling                                                                                                                                               
  File "usb/core.py", line 1292, in device_iter                                                                                                                                                            
  File "usb/backend/libusb0.py", line 456, in enumerate_devices                                                                                                                                            
  File "ctypes/__init__.py", line 369, in __getattr__                                                                                                                                                      
  File "ctypes/__init__.py", line 374, in __getitem__                                                                                                                                                      
AttributeError: /home/sbee/Downloads/ENLIGHTEN-4.0.12/libusb-1-150b88da.0.so.0.1.0: undefined symbol: usb_find_busses
```

Removing local dynamic libraries `./libusb-1.0*` prevents Enlighten from starting with a 'No backend available' error even though libusb-0*.so is present.

```
(base) sbee@sbee-ubu:~/Downloads/ENLIGHTEN-4.0.12$ ls .trash
libusb-1.0.so.0  libusb-1-150b88da.0.so.0.1.0
(base) sbee@sbee-ubu:~/Downloads/ENLIGHTEN-4.0.12$ ls | grep libusb
libusb-0-266a1141.1.so.4.4.4
```

samiebee43, 2023-06-19:
Some context at the pyusb level (v1.2.1) usb/core.py:1298

```
    if backend is None:
        import usb.backend.libusb1 as libusb1
        import usb.backend.libusb0 as libusb0
        import usb.backend.openusb as openusb

        for m in (libusb1, openusb, libusb0):
            backend = m.get_backend()
            if backend is not None:
                _logger.info('find(): using backend "%s"', m.__name__)
                break
        else:
            raise NoBackendError('No backend available')
```

Here we can see that pyusb unsurprisingly prioritizes libusb1 over usb0. From my previous comment we can see that the global installation of libusb1 is not affecting this particular search path (because removing local usb1 caused it to break all the way down to the exception).

So the question that remains is why `libusb-0-266a1141.1.so.4.4.4` is invisible to `usb.backend.libusb0`

usb/backend/libusb0.py:196 gives some hints about the local library search path:
```
def _load_library(find_library=None):
    return usb.libloader.load_locate_library(
                ('usb-0.1', 'usb', 'libusb0'),
                'cygusb0.dll', 'Libusb 0',
                find_library=find_library
    )
```

It looks like the hyphen should be removed..

closed 2023-06-19 by samiebee43

mzieg, 2023-06-22:
> The solution was the purge libusb1 from the build VM's global install before running `make linux-installer`.

Is there anything worth changing in our build instructions or build script to clarify or enforce this, so it doesn't happen again?

samiebee43, 2023-06-23:
Great point. Added.