WasatchPhotonics/ENLIGHTEN #194: Pass only one context to constructors, unify BusinessObjects and Controller

samiebee43, 2023-06-27:
# Introduction

This issue basically encapsulates the main goal associated with the sbee-refactor branch. Enlighten:main tends to pass a lot of elements from `ctl` separately into constructors.

There are a number of downsides to this.

1. Instantiations are long and obfuscated.
2. Some classes, such as Graph, are instanced more than once, the instantiator must keep track of many references to do so.
3. There are extra steps associated with accessing a different instance (such as multispec) or new UI widgets. 
4. References are renamed within the class body: makes searching, static analysis, and tracing more difficult.
5. Class initialization has a large block of code at minimum to perform the renaming, which distracts from the actual init.
6. Dependency cycles are more of a problem since references must be defined at init time as opposed to event time.

# Static Analysis / Complexity Reduction

For example, this is the instantiation of Graph in BusinessObjects.
 
https://github.com/WasatchPhotonics/ENLIGHTEN/blob/8fc41a6760c7dc6987b3556e2b13f603739ca7a3/enlighten/BusinessObjects.py#L187-L202

Instantiation is simpler with a single context object passed in.

https://github.com/WasatchPhotonics/ENLIGHTEN/blob/11bc4f6566990c5737ba8cb001fab26efe5b3ea8/enlighten/Controller.py#L245

Referencing widgets in Graph on main:

https://github.com/WasatchPhotonics/ENLIGHTEN/blob/8fc41a6760c7dc6987b3556e2b13f603739ca7a3/enlighten/scope/Graph.py#L113-L119

Referencing widgets in Graph on sbee-refactor:

https://github.com/WasatchPhotonics/ENLIGHTEN/blob/11bc4f6566990c5737ba8cb001fab26efe5b3ea8/enlighten/scope/Graph.py#L74-L80

In this case verbosity went up, but we have the advantage that we can see exactly where and how a widget is used given the name from within enlighten_layout.ui

```bash
$ grep -rn . -e 'ctl\.form\.ui.pushButton_zoom_graph'
./enlighten/scope/Graph.py:79:        self.ctl.form.ui.pushButton_zoom_graph.clicked.connect(self.toggle_zoom)
./enlighten/scope/Graph.py:227:        self.ctl.gui.colorize_button(self.ctl.form.ui.pushButton_zoom_graph, self.zoomed)
```

# Instance Reference Flexibility

The most important part is being able to reference other instances without having to (1) modify the constructor parameters (2) create a rename (3) modify the instantiation parameters first. This can be shown with GainDBFeature accessing an instance of Configuration:

https://github.com/WasatchPhotonics/ENLIGHTEN/blob/c36080868a00ac68fc83af6a6fb0c37cd33536c9/enlighten/scope/GainDBFeature.py#L129

And with EnlightenPlugin accessing novel widgets in one line per:

https://github.com/WasatchPhotonics/ENLIGHTEN/blob/8fc41a6760c7dc6987b3556e2b13f603739ca7a3/pluginExamples/EnlightenPlugin.py#L57-L63 

# Examples of refactor

In some cases you may want to do a deep refactor, in other cases you will want to make minimal changes. Here are some examples.

[GainDBFeature original](https://github.com/WasatchPhotonics/ENLIGHTEN/blob/main/enlighten/scope/GainDBFeature.py)
[GainDBFeature light refactor](https://github.com/WasatchPhotonics/ENLIGHTEN/blob/sbee-persist-gain/enlighten/scope/GainDBFeature.py): just adds `self.ctl` for easy access to other instances.
[GainDBFeature deep refactor](https://github.com/WasatchPhotonics/ENLIGHTEN/blob/sbee-persist-gain/enlighten/scope/GainDBFeature.py): uses `self.ctl` as only parameter, unwraps all ctl field renames.

Similarly, PluginController has some light refactors and a deep refactors. 

In this case, it would make sense to finalize and merge the deep refactor on GainDBFeature since it is fairly simple and not frequently modified. PluginController would be a sensible candidate to work slow, gradual changes. 

# Action

Ultimately, our decisions to include refactors will be limited by timing constraints--and it is best to make changes gradually over time. The objective here is to do full ctl unwrapping and single reference instantiation on simpler less critical classes. Simply adding ctl as the first parameter for other classes would be a good start.

Candidates for full ctl unwrap (deep refactor):
- [ ] scope/GainDBFeature
- [ ] file_io/LoggingFeature

Candidates for ctl parameter addition (light refactor):
- [ ] PluginController
- [ ] Measurement
- [ ] Multispec

Please check sbee-refactor, sbee-refactor-min to see if it's already been done.

# Scope

Once the five files listed above have been satisfactorily modified this issue can be closed. It would make sense to reference this issue in future issues about moving towards `self.ctl`.

samiebee43, 2023-06-27:
Working on GainDBFeature. First merged sbee-dev <-- sbee-persist-gain.

Now I would like to merge only GainDBFeature from sbee-refactor:

https://stackoverflow.com/a/64092840/1727470
```
git checkout <b1>
git checkout -b dummy
git merge <b2>
git checkout <b1>
git checkout dummy <path to file>
git branch -D dummy
```

mzieg, 2023-07-19:
> How and why was this ever labeled 'easy' ?

Heh.  I did Graph this morning as part of the ongoing ROI handles branch. That changeset is getting bigger than I expected or wanted...

mzieg, 2024-03-13:
Per BusinessObjects in 4.0.49, this has been done for 56 classes, and only the following 11 need done:

AccessoryControlFeature
AreaScanFeature
BLEManager
GridFeature
HardwareCaptureControlFeature
KIAFeature
LaserTemperatureFeature
PluginController
RamanShiftCorrectionFeature
ReferenceFeature
RegionControlFeature

mzieg, 2024-05-23:
Left to do as of 4.0.62:

* BLEManager
* AreaScanFeature
* RamanShiftCorrectionFeature
* AccessoryControlFeature
* RegionControlFeature