WasatchPhotonics/ENLIGHTEN #206: Discrepancy in file naming for 'unset' Prefix

samiebee43, 2023-06-28:
[image]

The prefix txtbox defaults to 'enlighten', but the naming scheme is different before and after the user touches that txtbox--even if they set it to match the default string 'enlighten'.

IMO a sensible correction would be to default the txtbox blank, and whenever it is blank, even after user input, use the simpler naming scheme. OR we can adopt a single naming scheme, likely the later including the prefix.

mzieg, 2023-06-28:
Is it possible that the nomenclature didn't change because you touched the Prefix textbox, but because you clicked the pencil icon to (potentially) edit one of the thumbnail labels?

I'm not disputing the potential of a bug, just trying to confirm the sequence of events to recreate.

samiebee43, 2023-07-03:
Yes exactly thanks for point that out.

Steps to reproduce are as follows:
1. Save spectra (notice the long form name in FS)
2. Click pencil to 'edit' name
3. Don't change anything, just press enter to confirm
4. Notice the name is in short form now

mzieg, 2023-07-04:
Right, so I don't think that's a bug.  You have "Rename files" checked in the Settings view.  That's what the behavior is supposed to do.  (This was a user-requested feature.)

closed 2023-07-04 by samiebee43