WasatchPhotonics/ENLIGHTEN #222: Line 92 of StatusBarFeature flagged by static analysis

samiebee43, 2023-07-03:
Line 92 flagged. SA believes that `menu.triggered` is a function and should not have a field `connect`. 

It seems like the intention was to do

`menu.connect('triggered', self.toggle_option_visible)`

or something similar, but I am only passing through this part of the code, and I am not sure what this is about

_Originally posted by @samiebee43 in https://github.com/WasatchPhotonics/ENLIGHTEN/pull/216#discussion_r1251296958_

mzieg, 2023-07-04:
Dunno, it's [definitely a signal](https://doc.qt.io/qt-5/qmenu.html#signals), and I can see that the callback (```toggle_option_visible```) is working (checking / unchecking items from the "..." menu on the StatusBar causes the respective fields to dis/appear).

samiebee43, 2023-07-04:
It's possible that PySide2 or Qt5 is using a __getattr__ as part of its bindings to Qt.
I'll consider this a 'reimplement', where the thing is working, but we'd be in better form if the analyzer can understand it.

closed 2023-12-13 by mzieg