WasatchPhotonics/ENLIGHTEN #276: RamanIntensityCorrection usage confusing

mzieg, 2023-09-12:
To enable RamanCorrectionIntensity, you have to switch to Expert Mode.
However, per the following code in Controller.py, you have to be in Raman Mode for the feature to actually be applied:
```
1823             ####################################################################
1824             # Raman intensity correction
1825             ####################################################################
1826                          
1827             if self.page_nav.doing_raman():
1828                 self.raman_intensity_correction.process(pr, spec)
```
We need to think how to do this more elegantly.

mzieg, 2023-09-13:
Related, we should add a Raman "Technique" in Expert Mode—that would also have resolved this, and the current list of exclusively non-Raman techniques is misleading.

samiebee43, 2023-09-13:
Relabeling as [ 1 - broken ] because the most intuitive way to use it fails silently. Though this is kind of the edge of these two labels.

Also added it to milestone _Planned for 4.0.16_. The idea with the "Planned for..." milestones is that we will not release until all of such issues have been closed. In exceptional cases we may move issues to subsequent versions, but be mindful to notify downstream users in that case.

We may prefer to do Planned for's for semi-major versions in the future, so we can still release small patches.

samiebee43, 2023-09-18:
Is this closed by #277 ? @mzieg

mzieg, 2023-09-18:
Hi Samie, I believe #277 adequately resolves this for now.

closed 2023-09-18 by samiebee43