WasatchPhotonics/ENLIGHTEN #291: enlighten.exe --help no longer works on Windows

mzieg, 2023-09-20:
Not sure when this broke, but isn't working now.  Also, `enlighten.exe --no-such-option` just silently fails, no message.

mzieg, 2023-10-07:
One of our recent changes fixed it:
```
C:\Program Files\Wasatch Photonics\ENLIGHTEN\Enlighten>enlighten --help
usage: enlighten [-h] [--log-level {debug,info,warning,error,critical}] [--log-append {False,True,LIMIT}]
                 [--logfile LOGFILE] [--max-memory-growth MAX_MEMORY_GROWTH] [--run-sec RUN_SEC]
                 [--serial-number SERIAL_NUMBER] [--set-all-dfu] [--stylesheet-path STYLESHEET_PATH]
                 [--window-state {floating,maximized,fullscreen,minimized}] [--plugin PLUGIN]

acquire from specified device, display line graph

options:
  -h, --help            show this help message and exit
  --log-level {debug,info,warning,error,critical}
                        logging level (default: info)
  --log-append {False,True,LIMIT}
                        append to existing logfile (default: LIMIT)
  --logfile LOGFILE     explicit path for the logfile (default: None)
  --max-memory-growth MAX_MEMORY_GROWTH
                        automatically exit after this percent memory growth (0 for never, 100 = doubling) (default: 0)
  --run-sec RUN_SEC     automatically exit after this many seconds (0 for never) (default: 0)
  --serial-number SERIAL_NUMBER
                        only connect to specified serial number (default: None)
  --set-all-dfu         set spectrometers to DFU mode as soon as they connect (default: False)
  --stylesheet-path STYLESHEET_PATH
                        path to CSS directory (default: None)
  --window-state {floating,maximized,fullscreen,minimized}
                        window initial state (default: floating)
  --plugin PLUGIN       plugin name to start enabled (default: None)
```
We should add the current version number to the output message, though.

samiebee43, 2023-12-05:
This has been fixed in 4.1.0 dev. The issue will close automatically once that is merged to main.

closed 2023-12-11 by mzieg