WasatchPhotonics/ENLIGHTEN #298: Improve usability on small screens

mzieg, 2023-09-26:
[image]

(That's 1920x1080 @ 150%, "Dell default")

samiebee43, 2023-09-27:
What other percentage scales are available on this system? I would defer to that if we can.

Otherwise we may look into UI Scaling as part of Qt. I know GTK has it's own scale parameter. That would prevent having to significantly redesign UI at the moment.

mzieg, 2023-09-28:
I changed the scale on that laptop to 100%, which made ENLIGHTEN much more usable.

This isn't high-priority, but I feel like relatively minor changes could improve usability in the odd case where we actually are being run at SVGA resolution.  I'm still playing with this in my mind. For instance, if form width < 1024, shrink thumbnail dimensions, maybe overlay Marquee over StatusBar, vertically shrink StatusIndicators etc. More of a journey than a destination, with no firm "hard spec" to reach. Just trying to be considerate to those with constrained resources :-)

closed 2023-12-14 by mzieg