WasatchPhotonics/ENLIGHTEN #299: "Turn Laser Off" button found not red

samiebee43, 2023-09-27:
> ENLIGHTEN BUG: I saw the "Turn Laser Off" on ENLIGHTEN when the button wasn't red...that button should always be red when the laser is firing (and it was definitely firing).  The StatusIndicators were all gray (disabled) as well.

samiebee43, 2023-10-17:
Need a unit with a laser to test and fix this.

Note this issue came before the new themes addition, but it would be worthwhile to make sure the button is still red while running the blue theme for example.

mzieg, 2023-10-17:
I just saw this on ENLIGHTEN 4.0.17 while/after using the BurnIn plug-in.

samiebee43, 2023-10-18:
[image]

mzieg, 2023-10-18:
I think this relates to run-BurnIn.bat running from C:\Users\foo\Documents\EnlightenSpectra\plugins\Prod, and therefore failing to load external relative CSS files. Problem was observed in ENLIGHTEN 4.0.17, probably already fixed in 4.1.0-dev.

mzieg, 2024-03-13:
closing for now, pending fresh observations