WasatchPhotonics/ENLIGHTEN #324: Load Measurement broken

samiebee43, 2023-10-12:
I was working on #108 when I noticed this. Attempting to load a measurement gives a "Loaded 1 measurement" but nothing shows up.

[image]

```
2023-10-12 17:11:18,304 [0x00004db8] enlighten.parser.ColumnFileParser ERROR    unable to convert to datetime: [['2022-09-29 15:59:28.481182']]
2023-10-12 17:11:18,304 [0x00004db8] enlighten.parser.ColumnFileParser DEBUG    active_pixels_horizontal = 1652
2023-10-12 17:11:18,304 [0x00004db8] enlighten.measurement.MeasurementFactory ERROR    failed to parse file C:/Users/sbee/Documents/EnlightenSpectra/New SIG Library/Methanol.csv
Traceback (most recent call last):
  File "Y:\projects\Enlighten-B\enlighten\measurement\MeasurementFactory.py", line 263, in create_from_file
    measurements = [ self.create_from_columnar_file(pathname, encoding=encoding) ]
  File "Y:\projects\Enlighten-B\enlighten\measurement\MeasurementFactory.py", line 410, in create_from_columnar_file
    return parser.parse()
  File "Y:\projects\Enlighten-B\enlighten\parser\ColumnFileParser.py", line 51, in parse
    self.post_process_metadata()
  File "Y:\projects\Enlighten-B\enlighten\parser\ColumnFileParser.py", line 204, in post_process_metadata
    state.boxcar_half_width = int(self.get_safe_float(key))
  File "Y:\projects\Enlighten-B\enlighten\parser\ColumnFileParser.py", line 89, in get_safe_float
    if len(v.strip()) == 0:
AttributeError: 'list' object has no attribute 'strip'
2023-10-12 17:11:18,305 [0x00004db8] enlighten.measurement.Measurements DEBUG    create_from_file: no Measurements parsed from C:/Users/sbee/Documents/EnlightenSpectra/New SIG Library/Methanol.csv
2023-10-12 17:11:18,305 [0x00004db8] enlighten.file_io.FileManager DEBUG    scheduling load of the next file
2023-10-12 17:11:18,306 [0x00004db8] enlighten.device.Multispec DEBUG    Multispec.current_spectrometer: self.device_id is None
```

This should be corrected and the surrounding try-except should make a msgbox upon failure instead of silently passing the traceback to the enlighten.log.

mzieg, 2023-10-13:
I ran into this a day or two ago, and am testing a change — not ready to call it a fix — in the ColumnFileParser changes on https://github.com/WasatchPhotonics/ENLIGHTEN/pull/313.

samiebee43, 2023-10-13:
[image]

I pulled that specific change into sbee-style and it worked for me. It allowed me to demonstrate the artifact I was looking for. #108 is resolved for the main graph but not for all the small ones.

samiebee43, 2023-10-13:
afaik this is resolved 4.1.0-dev.

mzieg, 2023-10-13:
> afaik this is resolved 4.1.0-dev.

I'm worried about how it happened though.  I haven't tried to trace through history, but I'd like to.

samiebee43, 2023-10-13:
It broke because of me on eea92bb. Really dropped the ball here. Sorry guys.

The commit message "fix broken CSV load" is concerning. I suspect I was testing on some file that needed scalar_metadata=False. I don't know what file that was, and I don't know what error I was getting at that time.

This time, the file I used to trace history, and open this issue, is attached. The error message is above `AttributeError: 'list' object has no attribute 'strip'`
[Methanol.csv - needs scalar_metadata=True](https://github.com/WasatchPhotonics/ENLIGHTEN/files/12899025/Methanol.csv)

---

If it breaks again, let's reopen this issue and be sure to attach the file and error message.

mzieg, 2023-12-11:
Fixed in 4.1.0

closed 2023-12-11 by mzieg