WasatchPhotonics/ENLIGHTEN #406: Build instructions do work "out of the box"

jkleinwp, 2024-05-13:
- Build instructions do not work "out of the box" on Windows 11 or Mac
- Not all requirements are captured and the user must manually install extra packages
- rebuild_resource.sh script works on Mac but not Windows
- BUILD_WINDOWS.md doesn't capture exactly what console tool should be used (e.g. Git BASH, command prompt, Power Shell); instructions/scripts do not work correctly in all these tools
-

mzieg, 2024-05-13:
rebuild_resources.sh does work on Win10/11 for me?

jkleinwp, 2024-05-13:
My results (Windows 11 Home):
- Git CMD console: 'sh is not recognized as internar or external command'. This is a fresh install of Git, so I must have to add sh executable to the path.
- 
- Git BASH: cannot find pyside executables in scripts/enlighten_py311 virtual environment.  The code that checks for the executable should be in double quotes to prevent globbing/word splitting (see [this](https://github.com/koalaman/shellcheck/wiki/SC2086))

Changing:
`(109) if ! [ -f $RCC ]`

to  
`(109) if ! [ -f "$RCC" ]`

Resolves this issue.  I think it's probably triggered by differences in our paths since my files are under a user directory that has a space in it.

PR #448 (2024-06-28): [#406] handle build directories with spaces in the path
Hoped to resolve #406